reivaxy/aquaMonitor #6: Monitor main power source

reivaxy, 2016-08-04:
Main power source should be monitored to send an SMS as soon as it fails, since the battery may not last long enough to wait for temperature to drop or, if it happens during lights off schedule.

Use a resistor divider bridge on main power (12V) to drive an arduino input.

NullFragment, 2016-08-04:
Are you asking for what resistors to use for the voltage divider?

In that case, using a normal voltage divider rule,

- R1 = 9.1k 
- R2 = 6.2k + 300 Ohm in series

Where

NullFragment, 2016-08-04:
Are you asking for what resistors to use for the voltage divider?

In that case, using a normal voltage divider rule,

- R1 = 9.1k 
- R2 = 6.2k + 300 Ohm in series

Where

- +12V is attached to side 1 of R1
- Arduino Power is connected to the junction of R1 side 2, R2 side 1.
- GND is attached to side 2 of R2

reivaxy, 2016-08-04:
Hi,
Thanks, no I know how to make the divider, it's just a reminder that I need to do it :)
I think I'll use slightly higher values though, to drain even less current.
But thanks again.

reivaxy, 2016-08-29:
Updated schematics and software, all is working nicely.

closed 2016-08-29 by reivaxy